ldocull/MaslowDue #2: 5v signal on JP3 pins 2&4

blurfl, 2019-03-18:
Looking at the schematic, I wonder whether the two pins on JP3 that are used to indicate the board version should be fed by 3v3 instead of 5v. I haven't hooked a Due up to a Maslow board, but the Arduino Due datasheets warn about damage from connecting 5v to the input pins. Have you found that it doesn't matter?

ldocull, 2019-03-18:
Good catch! I didn't experience any issue, but will update the schematic to reflect.  Those pins could be clipped off since they are not used. It would be cool to move the I2C to the hardware pins in future revs, although the accesses to EEPROM are few and far between, so not overly speed critical.    Thanks!

ldocull, 2019-03-18:
Updated data and notes posted. Thanks for your input!

closed 2019-03-18 by ldocull

aluminumwelder, 2019-03-18:
Thanks for doing this so fast. So do the pins need to still be clipped off? or are the gerber files now updated so no pins need to be clipped off?

ldocull, 2019-03-18:
Gerbers upload today have this adjustment.
Thanks,
Larry