ldocull/MaslowDue #3: does the firmware work with a linkage kit instead of ring?

aluminumwelder, 2019-03-23:
I noticed the read me file was very detailed with the motor distance and type of kit used. Are users with different motor spacing or linkage kits going to have issues? I am assuming they will just calibrate their machine like a normal maslow kit?

ldocull, 2019-03-23:
Currently there is no quadrilateral arithmetic to support linkage systems but it could be added. The current version only has triangular (ring) system support.

closed 2019-03-23 by ldocull

GeroBH, 2019-03-24:
Ring, top-mount and linkage are all triangular, just vary in the 'Rotation Radius'.
So the question is more if there is a setting that can be adjusted in radius.

ldocull, 2019-03-24:
There is no ring radius required. If the linkage system keeps the tool at the peak of the triangle then you’re all set. The posted firmware uses only the base and height of the total triangle (distance between motors and from that line to the tool, which is measured as the edge of the work surface) for its calculations.