ldocull/MaslowDue #9: Not really an issue...

terjeio, 2019-08-06:
I have ported grbl to a [number of 32-bit processors](https://github.com/terjeio/grblHAL/tree/master/drivers) that may be of interest. I am using a pointer based HAL to achieve a clean separation between the grbl core and the hardware. Recently I also added an option to the HAL for implementing different kinematics in a similar way. I have initially added (or rather ported) CoreXY and wallplotter code as a coding exercise for myself - but never tested it. Useful for this project?

BTW I stumbled over this while searching for CNC cards that uses LPC176x processors as I am currently rewriting the current incomplete driver. It is a bit disappointing that the pin allocations for the Smoothieboard and the Re-ARM board are suboptimal so I wondered if something better was to be found...

ldocull, 2019-08-15:
Good to know! You may be able to "diff" the latest Maslow-Due code against your base and see how it could be migrated in as a selectable option like CoreXY.   Keep up the good efforts!

closed 2019-08-15 by ldocull

terjeio, 2019-08-15:
Seemingly it can be migrated without changing the core grbl codebase, IMO it hooks nicely into the HAL. Will implement on a [TI Tiva C LaunchPad](http://www.ti.com/tool/EK-TM4C123GXL#) since it has two QEI modules capable of handling encoder pulses up to 20MHz without bothering the CPU. Thinking of using a stepper for Z instead of a DC motor.