ldocull/MaslowDue #18: A question..

TheOncomingStorm, 2022-02-18:
Does this firmware work with GroundControl/WebControl or is it Makerverse only?

closed 2022-02-24 by Maxattax97

TheOncomingStorm, 2022-09-15:
I hate to reopen this but I forgot to ask and this is relevant. For calibration I need to be able to nudge each individual motor until the sprockets' teeth line up to the "12 o'clock" so I can get accurate measurements. How can I do this? Ground/Webcontrol had that functionality but Makerverse didn't at all that I have managed to find and it drives me nuts. It took me 2 days(and I mean full 8+ hour days) to get Makerverse's calibration system to get me within a couple mm of accuracy, and I need sub-1mm accuracy or else I might as well sell this thing. bCNC doesn't seem to deal with this at all so I have to ask.

pelsound, 2022-09-20:
@TheOncomingStorm Please send your question via email to: support@makermade.com and we can help.
Thanks,
Joel